Nate711/Doggo #31: BEN

caobencheng, 2019-06-09:
Hi, guys, I am new here, and I have more than 25 years of microcontroller programming experience, and I found DOGGO platform on Internet, and I am very interested in developing more advanced features on this platform. Is any body knowing where can I or how can I buy a Doggo robot for testing? Thanks

Ben 
Petrel AI System, Surrey,Canada
my email: caobencheng@126.com

CrazyAZ, 2019-06-09:
We have no plans to produce and sell Doggo robots. The entire design is fully open-sourced on this github, so you would be able to purchase all the necessary parts, but all the assembly is DIY.

caobencheng, 2019-06-10:
Hi, Aaron,

Thank you for your reply. 

I would like to buy all the parts and assemble them by myself. So, where can I buy all these parts and how can I buy?

Regards

Ben

| |
曹本成
|
|
邮箱：caobencheng@126.com
|

签名由 网易邮箱大师 定制

On 06/09/2019 11:30, Aaron Schultz wrote:

We have no plans to produce and sell Doggo robots. The entire design is fully open-sourced on this github, so you would be able to purchase all the necessary parts, but all the assembly is DIY.

CrazyAZ, 2019-06-10:
The full project which contains the full CAD and other information can be found here: https://github.com/Nate711/StanfordDoggoProject

In it you can find the BOM which has a list of all the materials, which I can also link here: https://docs.google.com/spreadsheets/d/1MQRoZCfsMdJhHQ-ht6YvhzNvye6xDXO8vhWQql2HtlI/edit#gid=726381752

caobencheng, 2019-06-10:
Hi, Aaron,

Thank you for your fast reply. So, I have  some questions for you.
(1).  I can only order all the parts by myself from the vendors listed in the BOM?
(2). After I bought all the parts and assembly them, then I can download the software from your website to run the robot?
(3). As I am new here, do you have any recommendations for buying all the parts of DOGGO？

I will be very appreciate if you can give me some suggestions.

BEN

Petrel AI system, Surrey,BC, CANADA 

发自网易邮箱大师

CrazyAZ, 2019-06-11:
1) Yes, the only way to get the parts for Doggo is to buy them individually from the various vendors. However, some of the parts are generic enough that you can find them from vendors other than those we listed.
2) Yes, you will need to download and flash both the firmware for the Odrive motor controllers and the code for the main teensy. Both of these are available on the main project repo.
3) The greatest difficulty for acquiring the parts is the custom fabricated parts (3rd page of the BOM). I would definitely make sure you are able to acquire these parts before investing in the rest. We provide the CAD for each part, which you will likely need to outsource to a fabrication service. The services we used are linked in the BOM, but they may not be the best for you based on region.

caobencheng, 2019-06-11:
Hi, Aaron,

Thank you so much. I really appreciate your recommendations.

Ben

| |
曹本成
|
|
邮箱：caobencheng@126.com
|

签名由 网易邮箱大师 定制

On 06/10/2019 21:29, Aaron Schultz wrote:
Yes, the only way to get the parts for Doggo is to buy them individually from the various vendors. However, some of the parts are generic enough that you can find them from vendors other than those we listed.
Yes, you will need to download and flash both the firmware for the Odrive motor controllers and the code for the main teensy. Both of these are available on the main project repo.
The greatest difficulty for acquiring the parts is the custom fabricated parts (3rd page of the BOM). I would definitely make sure you are able to acquire these parts before investing in the rest. We provide the CAD for each part, which you will likely need to outsource to a fabrication service. The services we used are linked in the BOM, but they may not be the best for you based on region.

closed 2019-06-13 by Nate711

caobencheng, 2019-06-16:
Hi, Aaron,

I am right now starting to order the parts for the DOGGO robot. I found there is an important part which is “T-motor 5212 KV340” unavailable on amazon. Is there any other compatible part for the “T-motor 5212 KV350”? What is your suggestion for this?

Regards

Ben

Regards

Ben

| |
曹本成
|
|
邮箱：caobencheng@126.com
|

签名由 网易邮箱大师 定制

On 06/09/2019 11:30, Aaron Schultz wrote:

We have no plans to produce and sell Doggo robots. The entire design is fully open-sourced on this github, so you would be able to purchase all the necessary parts, but all the assembly is DIY.

CrazyAZ, 2019-06-17:
Replacing that motor with another would be fairly difficult, so I would look to buy the same one from a different source. Here are two other places you can find them:
http://store-en.tmotor.com/goods.php?id=377
https://www.foxtechfpv.com/tmotor-mn5212-kv340.html

caobencheng, 2019-07-13:
Hi, Aaron,

I have ordered all of the DOGGO parts. Now I found that the Odrive board should use 48V variant in your"README.md", but in your BOM list it says "
"ODRIVE , 24V, no connectors". So which is correct? I have ordered four 24V Odrive boards, can I use them or I need to replace 48V Odrive boards?

Regards.

BEN

发自网易邮箱大师

http://store-en.tmotor.com/goods.php?id=377
https://www.foxtechfpv.com/tmotor-mn5212-kv340.html

caobencheng, 2019-07-14:
Hi, Aaron,

I need some help from you. 

Today I start to compile the doggo code in PlatformIO IDE.There are some errors occurred as in the attachment picture. Do you know
what wrong with me?

I am using a Windows 10, installed Microsoft Visual studio 1.36.1, and PlatformIO IDE 2.2.0. I built my doggo project directory 
and copied the doggo code package and the ChRt package in the right directories. I opened the doggo project in the PlatformIO and 
opened the main.cpp, and then pressed the "BUILD" button. Then it started to compile, but at last some errors occurred.

I will be very appreciate if you can give me some advice for the problems as I am new here.

BEN

http://store-en.tmotor.com/goods.php?id=377
https://www.foxtechfpv.com/tmotor-mn5212-kv340.html

CrazyAZ, 2019-07-14:
Both the 24V and 48V Odrives will work. We list the 24V in the BOM because they are cheaper.

Unfortunately, I am not seeing any pictures, so I can't diagnose the error you are getting.

caobencheng, 2019-07-14:
Hi, Aaron,

Many thanks for your answer about the Odrive voltage. For the compiling errors, I attached the screen shot picture here. If you have some special concerns about the compiling and code burning, please let me known. 

Ben

n 7/14/2019 11:23，Aaron Schultz<notifications@github.com> wrote：

Both the 24V and 48V Odrives will work. We list the 24V in the BOM because they are cheaper.

Unfortunately, I am not seeing any pictures, so I can't diagnose the error you are getting.

caobencheng, 2019-07-16:
Hi, Aaron,

I am not sure if you got my email, so I resend it again.

Many thanks for your answer about the Odrive voltage. For the compiling errors, I attached the screen shot picture here. If you have some special concerns about the compiling and code burning, please let me known. 

Ben

n 7/14/2019 11:23，Aaron Schultz<notifications@github.com> wrote：

Both the 24V and 48V Odrives will work. We list the 24V in the BOM because they are cheaper.

Unfortunately, I am not seeing any pictures, so I can't diagnose the error you are getting.

CrazyAZ, 2019-07-17:
Hi Ben,

I am still not able to see any images. I think you need to send your reply on Github instead of through email in order for me to see it.

caobencheng, 2019-07-17:
Hi, Aaron,
Ok, I change to send the compiling error photo here so you can see it. Waiting for your reply. Many thanks.
BEN
![doggo_compile_errors]
(https://user-images.githubusercontent.com/51536550/61394289-9e2f8600-a877-11e9-827f-9feffb529a9a.jpg)

caobencheng, 2019-07-17:
Hi, Aaron,

I have sent the compiling error photo on Github. You can check it at:

https://user-images.githubusercontent.com/51536550/61394289-9e2f8600-a877-11e9-827f-9feffb529a9a.jpg

thanks.

Ben

发自网易邮箱大师

CrazyAZ, 2019-07-18:
That error is from code related to logging data to an SD card in the Teensy. I would first double check that you have the SdFat library properly initialized in the lib file. If you cannot find an issue there, I would simply comment out the datalogging code since it is completely non-essential to running Doggo.

caobencheng, 2019-07-18:
Hi, Aaron,

Many thanks for your reply. I would try to check if  the SdFat library properly initialized in the lib file. If I cannot solve the problem, then I will try to comment the datalogging code. I will let you known the result later. 

Ben

发自网易邮箱大师

caobencheng, 2019-07-21:
Hi, Aaron,

I have found the compiling problem and solved it. I found there were no contents in the subdirectories "lib\ChRt", "lib\SdFat",
and "lib\SparkFun_BNO080_Arduino_Library" when I downloaded the Doggo-master.zip package. So I further downloaded all of the 
missing contents of these three subdirectories. So now I can compile the project with no errors in PlatformIO. 

I thought the executive binary file is the "firmware.hex"(254K bytes) in the Doggo\.pio\build\teensy35 directory. So I burned 
it in the Teensy 3.5 board though the "Teensy loader" program. I though there should be USB_uart communication between the board 
and the computer. However, I could not monitor any communication information on the USB_uart port after reboot. 

As I could not found any guide for compiling, code burning, debugging information about the doggo code. So, I need to ask 
you the following questions.
(1). Do you have a guide for Doggo code compiling, code burning, debugging and Uart monitoring?
(2). Is the "firmware.hex" the final executive binary file?
(3). How can I monitor if the code is running? Will the LED on the board flash during running? 
(4). How can I use the USB_uart port to communicate with the board?

Best regards.

BEN

发自网易邮箱大师

CrazyAZ, 2019-07-22:
I'm glad you found the problem with the code compiling. As far as flashing the code to the Teensy that should all be handled by PlatformIO, we don't manually flash the hex file. Our code will cause the LED on the board to flash if it is running.  Lastly, to communicate and see if the code is running you can connect to the Teensy via USB and open up a serial monitor.  One of the easiest serial monitors to use is the one built in to the arduino application, but you can also find others.

caobencheng, 2019-07-23:
Hi, Aaron,

I tried to fix the problem of no LED flashing and no serial monitor output after uploading the code to Teensy 3.5 board.
But it still failed to see the doggo code running normally.

I did all the steps again from the start as the following:
(1). I connected the Teensy 3.5 board through an USB port of my Lenovo laptop computer. 
(2). On command line, I used "git clone --recursive https://github.com/Nate711/Doggo" to download the complete doggo code 
package.
(3). Then I ran "git submodule init" and "git submodule update" commands. It looks ok.
(4). In PlatformIO, I build the Doggo project. It looks the compiling has no problem. You can see the following compiling 
screenshot.

https://user-images.githubusercontent.com/51536550/61746107-d0495800-ad4f-11e9-97a9-1222e5d12f40.png

(5). Then I uploaded the code in the Teensy board in PlatformIO.  The "teensy loader" program was automatically popped out by 
the PlatformIO. You can see the upload screenshot as the following.

https://user-images.githubusercontent.com/51536550/61746149-e9520900-ad4f-11e9-9991-17ae3073f845.png

(6). I opened the serial monitor on the PlatformIO, but no serial message output. You can see the following screenshot.
I also used other UART monitor software, there was still no message output.

https://user-images.githubusercontent.com/51536550/61746172-fbcc4280-ad4f-11e9-814b-a1bb0010a099.png

(7). The LED light on teensy board was not flashing after the code was uploaded and rebooted. You can see the 
following photo.

https://user-images.githubusercontent.com/51536550/61746210-0edf1280-ad50-11e9-97e0-5b1417380b23.jpg

So, I am just wondering where am I wrong? 

Now the Teensy board is only connected to computer. Does the doggo code only run normally(LED flashing and serial message 
output) only when Teensy board is mounted on breadboard and connected to the Odrive boards?

I will be very appreciate that you can give me some ideas or hints so that I can make the doggo run asap. 

Best regards.

BEN

发自网易邮箱大师

caobencheng, 2019-07-24:
Hi, Aaron,

I have an idea. If you know the reasons of my problem, then just let me known. If you are not sure, can you send your 
Doggo code "firmware.hex" to me?  Then I can run your code on my Teensy 3.5 board to see if it can run normally.

Thanks.

BEN