Nate711/Doggo #35: Teensy pin 13 conflicted: BlinkThread & IMU SPI  all use it

kigerzh, 2019-07-26:
Dear Team,
Now I am rebuilding the Doggo . I check the code ,find that  "BlinkThread"  use the LED_BUILTIN  ( #define LED_BUILTIN   (13) ),    while the  IMU SPI port still use the pin 13 as the SCK . 
I know that I can re-assign a pin . But I just want to confirm that I start from a golden point .

CrazyAZ, 2019-07-27:
You are correct.  In the [config.h file starting on line 44](https://github.com/Nate711/Doggo/blob/2e710121f618b006577a700c81c319086a46a366/src/config.h#L44) you can see that we have reassigned the pins for the IMU to not include pin 13.

closed 2019-07-27 by CrazyAZ

kigerzh, 2019-07-28:
Dear Aaron, 
Thanks for your quick reply . 
I have found that code in the" config.h " .   That only  reassign the CS pin , Infact  the conflicated pin is the SCK pin .   
I checkd the BNO080 IMU board the SPI interface J6 need pins:   SCK, MISO, MOSI , !CS,WAKE , !RST  ;Tthe !INT on J7.

trebladev, 2021-06-17:
dear kigerzh,
i have same problem just like yours.can i ask you how you solve at last?
thanks

kigerzh, 2021-06-21:
> dear kigerzh,
> i have same problem just like yours.can i ask you how you solve at last?
> thanks

Hi  trebladev,

you can use the other pin such as pin 27 to work for SPI SCK.
for example: 
in the config.h 

// Pins for BNO080 IMU
#define SPI_CS_PIN 15
#define SPI_SCK_PIN 27  //13,14,27,
#define SPI_MOSI_PIN 11
#define SPI_MISO_PIN 12

in the imu.cpp

...

BNO080 bno080_imu;
float raw_integrated_gyro_y = 0;
 
THD_WORKING_AREA(waIMUThread, 4096);
 
THD_FUNCTION(IMUThread, arg) {
    (void)arg;
 
    // Turn on SPI for IMU
    SPI.setSCK(SPI_SCK_PIN); //Kiger Zhang

...