Nate711/Doggo #36: Doggo code seems not running

caobencheng, 2019-07-27:
Dear team,
I compiled and uploaded the Doggo code into my Teensy 3.5 board in PlatformIO, however there was no LED flashing and no serial information output for monitoring, so I donot know if the Doggo code is running. I spent many days and still not solved the problem. Did anybody meet this kind of problem and know how to solve it? Or, any special concerns needed for compiling the Doggo 
code? Thanks. BEN

CrazyAZ, 2019-07-27:
The LED will actually not start flashing until you have established a serial connection with the Teensy, so it is possible that the code is running but you have simply not been able to connect to it.

The first thing I would try is making sure that in config.h the line `#define USE_XBEE` is commented out. If it is not, the Teensy will be looking for communication from the XBee rather than the USB. Also make sure that your serial monitor is running at a baud rate of 115200.

caobencheng, 2019-07-27:
Hi，Aaron, thank you for your reply. Now the teensy board starts to run after I comment the #define USE_XBEE, and it can output serial information for monitoring. However, the LED still not flash. And the serial output "IMU beginSPI failed (non-fatal)...". How can I enable the LED flashing? Why the SPI failed? Thanks. BEN

caobencheng, 2019-07-28:
Hi, Aaron, I found the reason why the LED is not flashing after running. When the IMU thread fails, it will change the LED port to input type, so the LED will not flash. So, now I move the calling of BlinkThread after the calling of IMUThread in chSetup() , then the LED flashes normally.

CS4Liu, 2019-07-31:
> Hi, Aaron, I found the reason why the LED is not flashing after running. When the IMU thread fails, it will change the LED port to input type, so the LED will not flash. So, now I move the calling of BlinkThread after the calling of IMUThread in chSetup() , then the LED flashes normally.

Hi, thanks for sharing your solution of the LED flashing problem. I spent much time there too. 
But now, I meet another problem: when I send orders using USB Serial port, it  seems the corresponding thread is working from the output in the Serial Monitor, but the motor has no action!
Do you have this problem too?
And I test on one motor, I connect teensy pin9 to Odrive GPIO1, pin10 to GPIO2. Is this right?
And I don't  re-flash the ODrive firmware because the ODrive3.5 is perflashed with firmware.

caobencheng, 2019-07-31:
Thank you for sharing your experience on debugging doggo. We may share more and discuss more on this project. Right now I just start to assembly the system, and I right now can not answer your question, but maybe several days later when I make progress then I can share with you. Ben

| |
曹本成
|
|
邮箱：caobencheng@126.com
|

签名由 网易邮箱大师 定制

On 07/30/2019 19:17, cs.Liu wrote:

Hi, Aaron, I found the reason why the LED is not flashing after running. When the IMU thread fails, it will change the LED port to input type, so the LED will not flash. So, now I move the calling of BlinkThread after the calling of IMUThread in chSetup() , then the LED flashes normally.

Hi, thanks for sharing your solution of the LED flashing problem. I spent much time there too.
But now, I meet another problem: when I send orders using USB Serial port, it seems the corresponding thread is working from the output in the Serial Monitor, but the motor has no action!
Do you have this problem too?
And I test on one motor, I connect teensy pin9 to Odrive GPIO1, pin10 to GPIO2. Is this right?
And I don't re-flash the ODrive firmware because the ODrive3.5 is perflashed with firmware.

CrazyAZ, 2019-07-31:
@CS4Liu, you need to re-flash the odrives with the firmware we provide or else it will not work with our code.  We modified the communications protocol in order to make it faster and also changed the closed-loop control code.

CS4Liu, 2019-07-31:
> @CS4Liu, you need to re-flash the odrives with the firmware we provide or else it will not work with our code.  We modified the communications protocol in order to make it faster and also changed the closed-loop control code.

OK, thanks very much. I missed this step.

CS4Liu, 2019-07-31:
Hi, I download the source code, make successfully and get the '.hex' file. But after I re-flashed, my ODrive is not be recognized when I start odrivetool. The screenshoot is like:
[image]

I follows the ODrive docs to find my serial number. And run "odrivetool --serial-number 'my_number' ",but it don't works.
So I 'make clean' and 'make' again ,reflash the ODrive again. But comes another problem:wait for ODrive all the time:
[image]
Do you know how to solve this?

SamitHashaN, 2019-08-01:
Hi @caobencheng,
I am also following the same project. Flashed the firmware to odrive. But having issues with the Closed loop control. axis1 motor spis at a higher speed. Is there any solution for this?
Thank you.
Samith

caobencheng, 2019-08-01:
Hi, Samith, 
I can not answer your question this moment because I am not yet coming to the closed loop control point. If I come to this point later and I can solve this problem, I will share with you. However before that you may consult other guys or Aaron. Ben

| |
曹本成
|
|
邮箱：caobencheng@126.com
|

签名由 网易邮箱大师 定制

On 08/01/2019 12:37, SamitHashaN wrote:

Hi @caobencheng,
I am also following the same project. Flashed the firmware to odrive. But having issues with the Closed loop control. axis1 motor spis at a higher speed. Is there any solution for this?
Thank you.
Samith

caobencheng, 2019-08-04:
Hi, Aaron,

I found I may have missed ordering the Sparkfun BNO080 IMU. I double check the Bom of Doggo project, and I have not found the Sparkfun BNO080 IMU listed in the BOM. Do I need to order the Sparkfun BNO080 IMU before I assembly the whole system？
Regards
BEN

CrazyAZ, 2019-08-04:
It's weird that it is not in the BOM. You can assemble the whole system without it and add it later. The only thing you need it for right now is doing a backflip.

closed 2019-08-04 by CrazyAZ