Nate711/Doggo #37: How to configure ODrive before connect it to teensy?

JacobShenHdu, 2019-07-29:
Hi, I want to ask How to configure ODrive before connect it to teensy? I try to configure ODrive as follows but the motor go crazy(Rotating very fast) sometimes, is there any problem in config?

% init:
odrv0.erase_configuration()
odrv0.reboot()
odrv0.config.brake_resistance=0
odrv0.axis0.motor.config.pole_pairs=11
odrv0.axis0.motor.config.resistance_calib_max_voltage=4.0
odrv0.axis0.encoder.config.cpr=2000
odrv0.axis0.motor.config.current_lim=10
odrv0.axis0.motor.config.calibration_current=10
odrv0.axis0.motor.config.pre_calibrated=False
odrv0.axis0.encoder.config.pre_calibrated=False
odrv0.axis0.controller.config.pos_gain=0.01
odrv0.axis0.controller.config.vel_gain = 5.0/10000.0
odrv0.axis0.controller.config.vel_limit=5000.0
odrv0.axis0.controller.config.vel_integrator_gain = 0
% motor calibration:
odrv0.axis0.requested_state=AXIS_STATE_MOTOR_CALIBRATION
odrv0.axis0.motor.config.pre_calibrated=True
% encoder calibration:
odrv0.axis0.encoder.config.use_index=True
odrv0.axis0.requested_state
odrv0.axis0.requested_state=AXIS_STATE_ENCODER_INDEX_SEARCH
odrv0.axis0.requested_state
odrv0.axis0.requested_state=AXIS_STATE_ENCODER_OFFSET_CALIBRATION
odrv0.axis0.requested_state
odrv0.axis0.encoder.config.pre_calibrated=True
% startup config
odrv0.axis0.config.startup_encoder_index_search=True
odrv0.axis0.config.startup_closed_loop_control=True
odrv0.save_configuration() 
% test motor
odrv0.axis0.requested_state=AXIS_STATE_CLOSED_LOOP_CONTROL %(1)
odrv0.axis0.controller.pos_setpoint = 1000 %(2)

The motor go crazy sometimes at (1) and sometimes at (2) . The encoder seem work properly,  I read the encoder data find that set_postion at positive 1000 but the motor ratate in the negative direction and it's speed is getting faster and faster. Is there any sequential requirement for connection  motor and ODrive connections?

CrazyAZ, 2019-07-31:
Unfortunately, this is a very hard problem to diagnose on here.  I don't really see any issue with what you are doing. You might try asking on the ODrive forum to see if anyone there knows the answer to your issue. https://discourse.odriverobotics.com/

JacobShenHdu, 2019-08-01:
> Unfortunately, this is a very hard problem to diagnose on here. I don't really see any issue with what you are doing. You might try asking on the ODrive forum to see if anyone there knows the answer to your issue. https://discourse.odriverobotics.com/

Thank you for your reply. I set odrv0.axis0.encoder.config.use_index = False, and the problem has not yet appear.

closed 2019-08-01 by JacobShenHdu

mengqlTHU, 2019-08-01:
@JacobShenHdu Have you successfully run ODrive with teensy?

SamitHashaN, 2019-08-01:
> Hi, I want to ask How to configure ODrive before connect it to teensy? I try to configure ODrive as follows but the motor go crazy(Rotating very fast) sometimes, is there any problem in config?
>
> % [...]

Hi @JacobShenHdu I have the same issue. What was the solution for this?

CS4Liu, 2019-08-01:
> Unfortunately, this is a very hard problem to diagnose on here. I don't really see any issue with what you are doing. You might try asking on the ODrive forum to see if anyone there knows the answer [...]

Hello, are the all configure orders you list necessary?

JacobShenHdu, 2019-08-02:
Hi, @CS4Liu @SamitHashaN , It's my pleasure to share this interesting project with you, I configure ODrive as follows(by odrivetool or a script):
#Set parameters
odrv0.config.brake_resistance=0
odrv0.axis0.motor.config.pole_pairs=11
odrv0.axis0.motor.config.resistance_calib_max_voltage=4.0
odrv0.axis0.encoder.config.cpr=2000
odrv0.axis0.motor.config.current_lim=10
odrv0.axis0.motor.config.calibration_current=10
odrv0.axis0.motor.config.pre_calibrated=False
odrv0.axis0.encoder.config.pre_calibrated=False
odrv0.axis0.controller.config.pos_gain=100
odrv0.axis0.controller.config.pos_gain=0.01
odrv0.axis0.controller.config.vel_gain = 5.0/10000.0
odrv0.axis0.controller.config.vel_limit=5000.0
odrv0.axis0.controller.config.vel_integrator_gain = 0
odrv0.axis0.controller.config.kp_theta = 80
odrv0.axis0.controller.config.kd_theta = 0.5
odrv0.axis0.controller.config.kp_gamma = 50
odrv0.axis0.controller.config.kd_gamma = 0.5

# motor calibration(the motor will the scream)
odrv0.axis0.requested_state=AXIS_STATE_MOTOR_CALIBRATION
odrv0.axis0.motor.config.pre_calibrated=True
odrv0.axis1.requested_state=AXIS_STATE_MOTOR_CALIBRATION
odrv0.axis1.motor.config.pre_calibrated=True

# Set startup parameters
odrv0.axis0.config.startup_encoder_offset_calibration = True
odrv0.axis0.config.startup_closed_loop_control = True
odrv0.axis1.config.startup_closed_loop_control = True
odrv0.axis1.config.startup_encoder_offset_calibration = True

odrv0.save_configuration()
odrv0.reboot()

And I have tested ODrive with position control mode and coupled control mode. You can also refer to setup_doggo.py.

JacobShenHdu, 2019-08-02:
> @JacobShenHdu Have you successfully run ODrive with teensy?

I have tested ODrive, but not connected them to teensy yet.

SamitHashaN, 2019-08-02:
> odrv0.axis0.controller.config.pos_gain=100
> odrv0.axis0.controller.config.pos_gain=0.01

Why the pos gain is set to 100 and then to 0.01?

JacobShenHdu, 2019-08-02:
> > odrv0.axis0.controller.config.pos_gain=100
> > odrv0.axis0.controller.config.pos_gain=0.01
> 
> Why the pos gain is set to 100 and then to 0.01?

Sorry, I copy these commands from setup_doggo.py and foggot to modify it. I think "odrv0.axis0.controller.config.pos_gain=100"  is superfluous. And same settings to axis1.