Nate711/Doggo #38: doggo codes don't run as well as expected

crane-warrior, 2019-09-23:
Hello there! I have done previous jobs to run doggo codes on teensy and Odrive boards which have had been tested in doggo-firmware,are working well.The thing is when I commanded a JUMP with "J;" in Arduino Serial Monitor, the motors were nearly like stucking(the fact is that they moved 5-8 degrees) with noise of heavy "cluck".I have tried to load up the leg but it didn't move at all.The wiring of motors and Odrive, Teensy to Odrives are exactly copied from the image you guys provided.
My guess is that I have to re-tune the gains of close-loop-control because the gains and setponits of each specific movement is settled according to source code and the paras maybe differ form hardwares.
Have you guys ever met this before?I am looking forward your sooner reply!Thank you!

crane-warrior, 2019-09-23:
It is said that PD gains are different before unloaded and loaded so shall I test with the leg loaded?How did you guys tune those paras before?

crane-warrior, 2019-09-25:
hello,I have made some minor edit on the firmware and now it's avaliable for me to tune on the serial monitor.By the way I use the odrivetool "liveplot" to check the position feekback of both motors in one single odrive board.My idea upon tuning is to seperate one movement by time.  e.g, Jump is divided into 3 phases and in each phase the motors spin between two "position dot".
What came up next is really confusing.I have tune the pd gains(in which I kept the kp much lower and kd unchanged) until the position feedback is nearly like a square wave.However the range motors spined(position change),around 50-60 while the cpr is 2000 ,is too narrow to drive the leg.The setpoints shall not be altered with hardware-change So it seems like the soure codes you provided on the GitHub is wrong in someway.
Can you help me?Thank you.

crane-warrior, 2019-09-25:
I just check the angle motors spined,which is alpha and belta,are both about 3 degree.....strange...

Nate711, 2020-02-25:
Hi crane-warrior, were you able to resolve that issue? It seems like you had a problem with incorrect firmware. Our firmware has the capability to change the odrive gains by sending commands in the format of "g [theta kp] [theta kd] [gamma kp] [gamma kd]"