Nate711/Doggo #41: Hi, Doggo team, I have finished my doggo robot, and it can right trot, jump, hop, and dance. However, it can not walk. When I send "W" command, it just move at the same location and not go forward. I check https://github.com/Nate711/Doggo , it says that  "'W': (W)alk. Does not work currently". However I watch the youtube for the doggo project, in the video, it can walk. So, I just want to ask if the walk function included in the doggo code?

caobencheng, 2019-12-26:


CrazyAZ, 2019-12-27:
Doggo can indeed walk. The gait parameters we have the walk set to in the code right now give it a step length of 0. If you want to change it to walk forward you can do that here: https://github.com/Nate711/Doggo/blob/838562cd2474e398507204237319bdc6cec725fe/src/position_control.cpp#L106

Changing the 5th value to something like 0.1 will make the walk move forward.

closed 2019-12-27 by CrazyAZ

xuankaicheng, 2020-03-26:
> _No description provided._
i also have build a doggo，but the doggo can't stand up. set the parameters, put doggo on the floor
the leg do not stand horizontal，the doggo can not stand up what should i do？ thanks（mailbox ：kai540693711@163.com）